adafruit/Adafruit_Motor_Shield_V2_Library #14: Problem with microstepping code

cjones7, 2015-06-15:
I have a problem in the Motor shield V2 library.  Single, double, interleaved seem to function ok but 
the code for microstepping has an error.
It is in the step() function . Specifically the while loop, below, which is executed when the calls to 
onestep have finished.

if (style == MICROSTEP) {
    while ((ret != 0) && (ret != MICROSTEPS)) {
      ret = onestep(dir, style);
      delay(uspers/1000); // in ms
      steppingcounter += (uspers % 1000);
      if (steppingcounter >= 1000) {
	delay(1);
	steppingcounter -= 1000;
      } 
    }
  }

Considering the FORWARD direction only (BACKWARDS will be the mirror image).
ret is equal to currentstep.  Currentstep is constrained to 0 to 63 for either direction using the 
default 16 microsteps via the statements:

currentstep += MICROSTEPS*4;  
currentstep % MICROSTEPS*4;

If a single step had been executed (as in step(1,FORWARD, microstep)) then ret = 16 = MICROSTEPS and the while loop is exited.
If another single step is executed then ret = 32 and a further 32 calls to onestep are executed until 
ret = 0.
So we get with multiple calls to step but a single step: 1 step, then 3 steps, then 1 step then 3 
steps, and so on.
If 2 steps is repeatedly called then 4 steps will be executed, 3 steps called then again 4 steps 
executed, 4 steps called will execute the 4 steps.

So we have:
4n step move will give no extra steps, all ok.  So steps of 4,8,12,16,20,24,28,... will function 
correctly
4n+1 step repeated move will give the pattern of 1step, 3steps (1 step plus 2 extra), 1step, etc due to the unique position of being able to satisy the (ret != MICROSTEPS) condition of the while loop.
4n+2 step move will give 2 extra steps
4n+3 step move will give 1 extra step

For a 200 step motor 360 and 180 degree moves are ok as the moves are multiples of 4.  To see the worst case, move 200 single steps, the motor will do two revolutions not one.

Deleting the block of code removes the problem.  The code checks for a condition (currentsteps value returned with a value between 1 and 15) which would not be possible given the constraints on currentstep variable set within the onestep function.

If one wanted to keep the code (it does seem redundant to me) then the while condition could be:
  while ((ret != 0) && (ret % 4 < MICROSTEPS)) 

Regards,
CFJ

ladyada, 2015-06-17:
huh! we will try to replicate this and get back to you...its been a while since we looked at this code :)

tdicola, 2015-06-19:
Ah wow, yeah I set things up and saw doing 1 200 step call did a full 360, but 200 1 step calls did two full revolutions.  What you found with the need not to do the extra steps seems like it's spot on.  When I pulled out that while loop block both the single 200 step and 200 1 step calls worked great and did one revolution.  I just submitted the fix--thanks so much for digging in to find the issue!

commit 89324713 (2015-06-19): Remove extra microstep logic to fix #14.
closed 2015-06-19 by tdicola