adafruit/Adafruit_Motor_Shield_V2_Library #15: Shouldn't INTERLEAVE double steps?

jinschoi, 2015-07-21:
INTERLEAVE vs. MICROSTEP is handled inconsistently. INTERLEAVE gives one "step" per step, even though each "step" only gives half the motion. MICROSTEP considers the steps argument as full steps. Shouldn't steps be doubled in line 223 of Adafruit_MotorShield.cpp so that the steps argument is always treated as a full step?

ladyada, 2015-07-21:
its a design decision, what a 'full step' changes for INTERLEAVE because, well, it interleaves between single and double steps.

jinschoi, 2015-07-21:
Fair enough, although it leaves room for confusion when trying out the various step modes and only INTERLEAVE gives you half the travel of all the others.

closed 2021-09-21 by caternuson