adafruit/Adafruit_Motor_Shield_V2_Library #16: AFMS.begin(); never returns

audetto, 2015-08-12:
Hi,

I've just got my Arduino UNO and the Adafruit Motor Shield.
I've installed standard headers, Power to the Motor Shield (4 AA batteries about 5.4V) and 1 motor in position M3.

Then I try DCMotorTest and it prints only 

Adafruit Motorshield v2 - DC Motor test!

The motor never runs (I changed the code to use M3)

I tried to add prints in setup() after AFMS.begin() but they never trigger.

What could be happening?

jbcoulter, 2015-08-12:
I am about to get back to coding with the motor shield so I might not be the best one to respond as I am rust-EE!
Was the i2c identified in your sketch?

audetto, 2015-08-13:
What does it mean?
I've just copied the example DCMotorTest and canged 1 -> 3 for the dc motor to use.

audetto, 2015-08-17:
Probably had bad connections. Works well now.

closed 2015-08-17 by audetto