adafruit/Adafruit_Motor_Shield_V2_Library #22: Stepper motor release hardware bug

kb1isp, 2016-04-29:
This may not be a typical use case but I have noticed the tendency to have an extra step when shutting a stepper down with the release function. I have traced this to the sequencing of the shutdown of the PWM. It occurs after the enable lines are toggled. This continues to modulate the windings as the enables are being released.

caternuson, 2021-09-21:
Would moving the `setPWM` calls before the `setPin` calls fix this?
https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library/blob/64508284e5636f69e7dfae909e73d32f5487136f/Adafruit_MotorShield.cpp#L296-L303

kb1isp, 2021-09-22:
Hi Nelson,

Thanks for looking at this. I haven’t worked on it for a while, but as I
recall this made the most sense for me at the time. I had some stored flux
in the device, and combined with the timing, that was causing some trouble.

At the time, it just made sense to set everything to a known state before
proceeding. Perhaps PwM setting accomplishes the same thing, but this was
just more apparent and expedient for me. Certainly if you think it’s better
for your application then go for it.

Also, recently haven’t been able to find this chipset easily. If you know
anything I’d be interested.

Regards,
Luke

caternuson, 2021-09-22:
Hey! Thanks for responding. Yah, sry, it's an old issue. Trying to do some clean up in this repo.

I think let's just close this for now. We can reopen it if you run into this again. Just reply back here. That way we'll have something to test against. I'd think even using a faster processor might change the behavior as well.

closed 2021-09-22 by caternuson

kb1isp, 2021-09-22:
Sure, no worries. This was in a past life for me anyway, lol. As I recall
it was actually the motor damping not being taken into account so the thing
wouldn’t settle.

Thanks again,
Luke