adafruit/Adafruit_Motor_Shield_V2_Library #23: WDT Error and following automatic Reset

rudydevolder, 2016-08-11:
- Arduino board:  **Feather Huzzah**

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.10

- List the steps to reproduce the problem:
Just running the standard **steppertest.ino** example 

I found the solution already see: [https://github.com/esp8266/Arduino/issues/2377 ]
By adding in the library-code the delay(0) command.:
*
```
  while (steps--) {
    //Serial.println("step!"); Serial.println(uspers);
    ret = onestep(dir, style);
    delayMicroseconds(uspers);
	//Rudy De Volder: to solve WDT problem with the ESP8266 simply add the following line:
	delay(0);
  }
```
*

karelv, 2016-12-05:
A similar issue on my Feather Huzzah / motor shield is present. I hope your patch fits also my HW!

According to:
https://github.com/esp8266/Arduino/blob/master/doc/reference.md#timing-and-delays
you might also consider to replace the 
```
delay (0);
```
by
```
yield();
```
I will try later today... and let you know.

ladyada, 2016-12-05:
ok added a yield, please tr it!

https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library/commit/a315ff70884e3c43a144b522d67d61267bd4cf6d

closed 2016-12-05 by ladyada

karelv, 2016-12-05:
I checked both `delay (0);` and `yield ();` ==> both are working for me.
Thanks for adding the `yield` function.