adafruit/Adafruit_Motor_Shield_V2_Library #35: Please expose existing "all on or off" functionality

Jerware, 2019-04-06:
- Arduino board:  Uno
- Arduino IDE version:  Any

The Motor Shield library includes the following 'all on or off' functionality for the motor controllers.

`/**************************************************************************/
/*!
    @brief  Helper that sets the PWM output on a pin and manages 'all on or off'
    @param  pin The PWM output on the driver that we want to control (0-15)
    @param  value The 12-bit PWM value we want to set (0-4095) - 4096 is a special 'all on' value
*/
/**************************************************************************/
void Adafruit_MotorShield::setPWM(uint8_t pin, uint16_t value) {
  if (value > 4095) {
    _pwm.setPWM(pin, 4096, 0);
  } else 
    _pwm.setPWM(pin, 0, value);
}`

However, there's no way to access this feature from the sketch level. The public setSpeed function, which calls setPWM, only accepts a uint8_t.

`void Adafruit_DCMotor::setSpeed(uint8_t speed) {
  MC->setPWM(PWMpin, speed*16);
}`

255 * 16 = 4080, so the user can't get 4096 to setPWM. This functionality should be exposed for rare cases when PWM causes too much electrical noise and 12V DC is all that's needed.

ladyada, 2019-04-07:
hiya please submit a PR and we'll take a look!

caternuson, 2021-09-21:
Done with #44. See the new `fullOn()` and `fullOff()` methods.

closed 2021-09-21 by caternuson